vsergeev/lua-periphery #1: Version 1.0.3 on raspbian - compile error

1linux, 2015-01-25:
Current github version does not compile on raspbian / Lua 5.1
First you have to set include-path before make: `export LUA_INCDIR=/usr/include/lua5.1` 
Then when compiling:
> src/lua_i2c.c: In function ‘luaopen_periphery_i2c’:
> src/lua_i2c.c:371:5: error: ‘I2C_M_STOP’ undeclared (first use in this function)
> src/lua_i2c.c:371:5: note: each undeclared identifier is reported only once for each function it appears in
> Makefile:40: recipe for target 'periphery.so' failed

vsergeev, 2015-01-26:
What kernel version are you building for?

1linux, 2015-01-30:
I tried again on a BananaPi, same issue...
Version:
>uname -a
>Linux lemaker 3.4.90 #2 SMP PREEMPT Tue Aug 5 14:11:40 CST 2014 armv7l GNU/Linu

>cc -Wall -Wextra -Wno-unused-parameter  -fPIC -I. -I/usr/include/lua5.1 -shared src/lua_periphery.c >src/lua_mmio.c src/lua_gpio.c src/lua_spi.c src/lua_i2c.c src/lua_serial.c c-periphery/periphery.a -o periphery.so
>src/lua_i2c.c: In function ‘luaopen_periphery_i2c’:
>src/lua_i2c.c:371:5: error: ‘I2C_M_STOP’ undeclared (first use in this function)
>src/lua_i2c.c:371:5: note: each undeclared identifier is reported only once for each function it appears in
>Makefile:40: recipe for target 'periphery.so' failed
>make: *** [periphery.so] Error 1

When I define `I2C_M_STOP` manually it works... crazy.

vsergeev, 2015-01-30:
Thanks. It appears `I2C_M_STOP` was added in kernel version 3.6. I'll add some #ifdefs so lua-periphery will only export the flags available in the kernel headers it's built against. That should fix your issue and make the build more robust.

closed 2015-01-30 by vsergeev

vsergeev, 2015-01-30:
I've released v1.0.4 with this fix and updated moonrocks with the new rock. Thanks again for reporting this.

commit 5149ca5c (2015-01-30): only export I2C_M_STOP constant in i2c if it's available in linux headers

this fixes compilation on kernels older than 3.6, which don't support this I2C
message flag.

resolves #1. thanks to 1linux for reporting this.