vsergeev/lua-periphery #5: Adjust build system to support an external c-periphery

tpetazzoni, 2015-05-16:
Currently, the biuld system of lua-periphery automatically clones and builds the c-periphery library by itself. While this is probably practical for some users, it is not practical at all for embedded Linux build systems such as Buildroot.

Such build systems really like to ensure that they download all the source code they build, in order to guarantee that the system can be rebuilt in the exact same state from an offline (i.e not Internet connected) machine. What lua-periphery does is currently breaking this assumption.

Would it be possible to introduce an option to the Makefile to support using an already built c-periphery library?

You can look at http://patchwork.ozlabs.org/patch/473047/ for the two patches I've proposed to Buildroot to solve the problem. Of course, those patches break the current clone+build mechanism, so a better solution will be needed for upstream lua-periphery.

vsergeev, 2015-05-27:
Yes -- this must have been very strange to discover in the lua-periphery build process. Sorry about that. It turns out that it's also a problem for LuaRocks' packaging model: Rockspecs can be packaged into source rocks, which should contain all the necessary sources to build the rock, without further requiring an Internet connection. lua-periphery's additional git clone in its build process breaks that invariant, as it does in buildroot. Let's just say it's not something I'm proud of.

The issue was that c-periphery was too small of a project to warrant packaging for all major distributions (which would have also been a major hassle of its own), and so it was unreasonable for lua-periphery to depend on it in that way. lua-periphery is really just Lua bindings to c-periphery, so the git submodule approach was the easiest way to express that dependency. However, LuaRocks' git fetch helper doesn't support recursive cloning -- a known limitation -- and there is no way to specify additional fetch-stage commands, so the LuaRocks infrastructure left very few options. Among them was this clone-during-build nonsense, which several other packages have also had to resort to. See more details [here](https://github.com/keplerproject/luarocks/issues/334#issuecomment-105180965).

The good news is that the `gitrec` fetcher contributed under that issue's comments works well, so I'm going to change the lua-periphery rockspec to depend on and use it once it's uploaded to LuaRocks. Then I'll update the lua-periphery package version on buildroot. This should resolve this issue.

As for keeping c-periphery as a package in buildroot -- we could leave it in there, in case other developers wish to leverage it. I could add an `install` target to its Makefile that installs the static library in the right place, so the `.mk` file won't have to. Otherwise, the c-periphery package is unnecessary with the fixed lua-periphery rockspec.

tpetazzoni, 2015-05-27:
Thanks a lot for the feedback. If I understood correctly, then this new gitrec thing should be OK for Buildroot.

Another possibility you have, if c-periphery is not that useful on its own, is to simply include the c-periphery code in lua-periphery, and get rid of c-periphery as a separate project.

This would for example avoid the situation in which you are today: lua-periphery is referencing an older version of c-periphery, which doesn't have certain fixes that we (Buildroot) contributed and were interested in (fixing the build on MIPS).

vsergeev, 2015-05-27:
Absorbing c-periphery into lua-periphery is certainly an option, but for the time being I think it may still provide some utility to userspace embedded Linux applications or to future language bindings. Time will tell.

The latest version of lua-periphery (v1.0.5) pulls in c-periphery (v1.0.3), which does contain the MIPS build fixes.

The buildroot lua-periphery package uses v1.0.4, while the buildroot c-periphery fetches commit id vsergeev/c-periphery@0fdfac3264f1dce32991f4ec30d084e34b35181d -- though this commit id does contain the MIPS build fixes, it also has a bug that breaks Serial functionality and wasn't fixed until the following commit vsergeev/c-periphery@c1474c97602035922af9ca6ad11a771cde96cb0d. The MIPS build fix and following Serial bug fix are included in c-periphery v1.0.3.

vsergeev, 2015-05-27:
The moment `gitrec` appears on LuaRocks, I will release a lua-periphery v1.0.5-2 rockspec package and update the buildroot package.

vsergeev, 2015-05-27:
Resolved in 397db999944fd4366cc51beaa710130c553b1884.

Patches updating relevant packages sent to buildroot mailing list.

closed 2015-05-27 by vsergeev