vsergeev/lua-periphery #6: local GPIO = require('periphery').GPIO returns nil

langley, 2015-09-18:
While working with lua 5.2 and a local build of lua-periphery the require is returning nil. 
If I have LUA_CPATH setup wrong, I get complaints about not finding the library. 

So it's as though the .so is being found, but for some reason it's not initialized properly. 

Thanks for any pointers, I'm very excited to use this library.

vsergeev, 2015-09-23:
Could you provide a log of the build and an example of how and where it is being imported relative to the built library?

langley, 2015-10-09:
Thanks for offering to get to the bottom of the problem. But I think "I was the bottom of the problem" ;^) Turns out that I was using the lua interpreter / REPL and I shouldn't have been using a 'local' there. 

Thanks again.

closed 2015-10-09 by langley