vsergeev/lua-periphery #10: Multiple GPIO reads fail

gw0, 2016-03-16:
As I understand GPIO ports, once opened, should be able to be read from or written to multiple times. But for reading only the first read works.

```
# lua
Lua 5.1.5  Copyright (C) 1994-2012 Lua.org, PUC-Rio (double int32)
> GPIO = require("periphery").GPIO
> gpio = GPIO(22, "in")
> print(gpio:read())
false
> print(gpio:read())
(error object is not a string)
>
```

vsergeev, 2016-03-26:
Hello gw0,

Sorry about just getting to this now, it appears I missed it in my feed.

That's correct, and the loopback test (https://github.com/vsergeev/lua-periphery/blob/master/tests/test_gpio.lua#L98) does do multiple reads in succession. This leads me to believe it might be related to your environment and how the GPIOs are being rewound.

Could you paste the output of `print(tostring(gpio:read())` on the second attempt? This will format the error object on Lua 5.1. Also, could you print the version of lua-periphery, with `=require("periphery")._VERSION`?

gw0, 2016-04-12:
Sorry, but I don't have access to the OpenWRT device with Lua where this was running.

Versions of `lua-periphery` and `c-periphery` were the ones you got by cloning it ~20 days ago and manually cross-compiling to mipsel architecture (see https://github.com/vsergeev/lua-periphery/pull/9).

vsergeev, 2016-09-29:
I'm going to close this issue for now -- feel free to re-open if you (or someone else) encounters this problem again, and we can investigate further.

closed 2016-09-29 by vsergeev