vsergeev/lua-periphery #18: Is it compatible with Lua 5.3 and Rasp Pi 3 B+

gustavohmsilva, 2018-11-13:
Hi, is it compatible with Is it compatible with Lua 5.3 and Rasp Pi 3 B+

drauschenbach, 2019-05-07:
I don't know about Lua 5.3, but I did just confirm that this module at least loads under Lua 5.2 on the Raspberry Pi 3 B+:

```sh
$ apt-get install lua5.2 lua5.2-dev
$ export LUA_INCDIR=/usr/include/lua5.2
$ make clean all
```

```lua
> I2C = require 'periphery'.I2C
> print(I2C.version)
1.1.3
> i2c = I2C('/dev/i2c-1')
> junk = I2C('/dev/nonexistent')
Opening I2C device "/dev/nonexistent": No such file or directory [errno 2]
```

gustavohmsilva, 2019-05-07:
Lua5.2 usually generate problems with much of the other libs around, that normally focus on being =5.1 or >5.2.

I had problems with 5.3 for this library, unfortunately.

vsergeev, 2019-05-08:
lua-periphery supports Lua 5.1 (including LuaJIT), Lua 5.2, and Lua 5.3. What problems did you have with 5.3?

vsergeev, 2019-10-04:
Closing for now, let me know if anyone has issues running on >= Lua 5.1. My regular test environment is with Lua 5.3 on a Raspberry Pi 3.

closed 2019-10-04 by vsergeev