vsergeev/lua-periphery #19: Lacking PWM feature

gustavohmsilva, 2018-11-14:
I've noticed that although the Python version has PWM capabilities, the Lua version don't count with it. Is this due to some limitation in the Language or just wasn't implemented? If so, is this something on a future milestone?

vsergeev, 2019-10-04:
No limitations -- the PWM and LED modules in python-periphery were user contributions. They can be added to c-periphery down the line (or sooner with a PR!), and then easily bound to lua-periphery.

I think I'll create issues here linked to c-periphery issues for each of the modules missing from c-periphery/lua-periphery, but present in python-periphery. It would be nice to have consistent support and API across all three.

closed 2019-10-04 by vsergeev

Janaka-Steph, 2019-12-13:
Sorry to reopen this issue, but I am also interested by this feature. Where can I follow the development? Any other recommendation to have PWM in Lua on Raspberry?

vsergeev, 2019-12-15:
I'm actually planning on implementing LED and PWM support in both c-periphery and lua-periphery in the coming 1.5 weeks, so there will be a v2.1.0 release of both soon.

I'm not aware of other Lua libraries with PWM support, but you could access the `/sys/class/pwm` files directly with the built-in `io` library in the meanwhile.

Janaka-Steph, 2019-12-15:
Ok great, thank you for your detailed answer.

vsergeev, 2020-01-08:
@Janaka-Steph I just released v2.1.0 of c-periphery and lua-periphery with LED and PWM support.

Janaka-Steph, 2020-01-08:
Woohooo thank you!!