vsergeev/lua-periphery #20: Serial read hang

ThreeMileJump, 2019-02-28:
If I attempt a read on the serial device with a timeout set, after pulling out the USB cable, then the read hangs.  If the USB cable has been pulled out, then the serial port should have closed and read should give an error, or at least time out and return no data.

The problem here is that closing the USB serial device seems to result in a hang that can only be fixed by closing the program. USB port closing down needs to be detectable so the user can do something about it.

Under linux, a brief USB disconnect will move the device to the next COM port - which is hard to detect and fix without error messages.

vsergeev, 2019-10-04:
I will look into recreating this locally. I would expect the underlying read() to fail with a EIO.

vsergeev, 2019-10-09:
Thanks for reporting. Fixed in c-periphery v2.0.1 / lua-periphery v2.0.1.

vsergeev, 2019-10-09:
This condition will return an error of `SERIAL_ERROR_IO` with the message `Reading serial port: unexpected empty read`.

closed 2019-10-09 by vsergeev