vsergeev/lua-periphery #21: Support for internal pull down resistors

Henkoglobin, 2019-12-14:
The Raspberry Pi has internal Pull Down resistors that can optionally be enabled ([This article](https://grantwinney.com/using-pullup-and-pulldown-resistors-on-the-raspberry-pi/#option-2-enabling-an-internal-pull-down-resistor-in-the-code) demonstrates how to do that in Python). 

I have not found any option to do this with lua-periphery... Is this an oversight in the library or am I missing something? Should it be implemented? Or is it just not documented?

Thanks in advance for any insights!

1linux, 2019-12-15:
Maybe this is for you:

https://github.com/1inux/lua-gpio-bcm2708

Henkoglobin, 2019-12-21:
Thanks, this seems very useful. I haven't tried it yet, but I will shortly after christmas.

However, I feel like this should be native functionality of lua-periphery, shouldn't it? Is there a reason why it's not supported out-of-the-box?

vsergeev, 2020-01-08:
@Henkoglobin I'll look into this for the periphery series.

vsergeev, 2020-03-27:
Unfortunately, this configuration is specific to Broadcom's GPIO controller and not exposed in the Linux userspace API for GPIOs, so it's a bit out of scope of the project. I would recommend just setting the pull ups/downs through the device tree overlay if your configuration is static, or using a wrapper library as @1linux suggested.

closed 2020-03-27 by vsergeev

Henkoglobin, 2020-03-27:
@vsergeev I see, thank you for looking into it!