vsergeev/lua-periphery #23: error in cross-compile by yocto generated sdk

nmarandi, 2020-04-05:
I am trying to cross-compile lua-periphery by the SDK generated by yocto for me.

but I face this error:

`make[1]: Entering directory '/home/milad/git/lua-periphery/c-periphery'
arm-angstrom-linux-gnueabi-gcc  -O2 -pipe -g -feliminate-unused-debug-types  -std=gnu99 -pedantic -Wall -Wextra -Wno-unused-parameter  -fPIC -DPERIPHERY_VERSION_COMMIT=\"v2.1.0\" -Wl,-O1 -Wl,--hash-style=gnu -Wl,--as-needed  -c src/gpio.c -o obj/gpio.o
src/gpio.c:10:20: fatal error: stdlib.h: No such file or directory
 #include <stdlib.h>`

I found the root cause. in the makefile, CC and AR are defined for CROSS_COMPILE but for yocto SDK, defining them is not needed because they are defined in its environment-setup file.

vsergeev, 2020-04-25:
I'm not sure I completely understand the issue or the solution. Does it build by unsetting CROSS_COMPILE? Is the wrong compiler being used in your error?

nmarandi, 2020-04-25:
CC = $(CROSS_COMPILE)gcc
AR = $(CROSS_COMPILE)ar

these values are set in your [makefile](https://github.com/vsergeev/c-periphery/blob/master/Makefile).

but for cross-compile, for **$(CROSS_COMPILE)gcc**, the **--sysroot=$SDKTARGETSYSROOT** need to be set.
for example in yocto generated SDK CC variable is like this:

```
CC="arm-angstrom-linux-gnueabi-gcc  -march=armv7-a -mthumb -mfpu=neon  -mfloat-abi=hard --sysroot=$SDKTARGETSYSROOT"
```

vsergeev, 2020-04-25:
I see. In general, the appropriate flags and variables will depend on your target and build environment. The Makefiles in both lua-periphery and c-periphery are designed to append to variables like CFLAGS or LDFLAGS to be flexible with different build environments. In your case, it seems the you need to pass the sysroot in the CFLAGS, in addition to selecting the cross-compiler, so the invocation may also look like:

```
CROSS_COMPILE="arm-angstrom-linux-gnueabi-" CFLAGS="-march=armv7-a -mthumb -mfpu=neon -mfloat-abi=hard --sysroot=$SDKTARGETSYSROOT" make
```
There isn't a general way to handle these details in a single Makefile. If the goal is simplifying the build for Yocto, it would probaby be best to create a Yocto build recipe that passes the right build variables to the Makefile.

vsergeev, 2020-04-26:
Closing for now. Feel free to reopen if you run into any other build issues.

closed 2020-04-26 by vsergeev