vsergeev/lua-periphery #25: how to send hex data with serial?

YoonesImani, 2020-10-03:
Hi, how can I send serial data in HEX format?

Thanks

vsergeev, 2020-10-03:
This is a bit vague. Do you want to send binary data? or binary data encoded into an ASCII hex string?

YoonesImani, 2020-10-10:
actually, I want to send binary data.

vsergeev, 2020-10-10:
Lua strings can contain binary data, so you can write binary data with the `write()` method on the serial handle (https://github.com/vsergeev/lua-periphery/blob/master/docs/serial.md).

closed 2020-10-10 by vsergeev

YoonesImani, 2020-10-11:
The problem is for example when I want to send \r (carriage return) it
sends two characters "\" and "r"

vsergeev, 2020-10-11:
I think your problem is somewhere in the byte string construction.

The Lua string "\r" is one byte:

```
Lua 5.4.0  Copyright (C) 1994-2020 Lua.org, PUC-Rio
> s = "\r"
> #s
1
> s == "\x0d"
true
> string.byte(s, 1) == 0x0d
true
> 
```