vsergeev/lua-periphery #26: sh: 1: make: not found when installing with LuaRocks?

j-hmd, 2021-01-08:
Hi! I'm new to Lua, and I was trying to install lua-periphery, but I get the following error:
Installing with: `sudo luarocks install lua-periphery` 
```
Installing https://luarocks.org/lua-periphery-2.3.1-1.src.rock
Warning: variable CFLAGS was not passed in build_variables
sh: 1: make: not found
```
Am I doing something wrong?

On Ubuntu20.04 Desktop.

The previous commands I ran to start were:
```
sudo apt-get update
sudo apt-get install lua5.1
sudo apt-get install liblua5.1-0-dev
sudo apt-get install lua-socket
sudo apt-get install luarocks
sudo luarocks install luasocket
```

Thanks!

vsergeev, 2021-01-08:
Is `make` and `gcc` installed? You probably just need `sudo apt-get install build-essential`.

j-hmd, 2021-01-09:
Oh, amazing! I had gcc, and thought I should have make. Thank you!

closed 2021-01-09 by j-hmd